gmlarumbe/tree-sitter-systemverilog #2: May you add some introduction?

yuhaos, 2024-05-28:
Hi,

I am search for tree-sitter plugin for verilog/systemverilog. "tree-sitter/tree-sitter-verilog" doesn't work well for highlighting.
I am wondering how to use "tree-sitter-systemverilog" in nvim. May you kindly add some introduction about this plugin and on how to use it? 

Thanks & Regards
Yuhao

zhangwwpeng, 2024-06-24:
Thanks to the author for the project. Recently i adapted to nvim. now  you can run :TSInstall systemverilog. but weak performce for nvim. I will optimize in the future

closed 2024-07-26 by gmlarumbe

yuhaos, 2024-09-08:
Hi @gmlarumbe ,

Thanks for the great work from you and @zhangwwpeng .
Sorry for the later reply, the solution works fines in NVIM.

Regards
Yuhao