gmlarumbe/tree-sitter-systemverilog #4: ; not included in node

henrykvdb, 2024-09-18:
Hey so I think I ran into an issue (unless I am interpreting wrong / there is something wrong about the nvim integration)

I have a `(generate_block)` node

[image]

which contains;

1-  name; `simple_identifier`

[image]

2- a `data-declaration`

[image]

3 - a `local-param-declaration`

[image]

**ISSUE: the `;` is not part of the local param declaration. In all other cases that I tested the `;` is part of the statement that it terminates**

4 - a `comment`, for sanity ;)

[image]

henrykvdb, 2024-09-18:
I think I wasted more words than necessary to say that ; is not a child of the (local)parameter that it belongs to.
The generate block is irrelevant to the issue, that's just how I noticed it. Sorry for complicating the issue

You can reproduce it with

```
module test;
localparam int MyNum = 5; // ; not part of node, strange
int alsoMyNum= 6; // ; part of noce, expected
endmodule
```

gmlarumbe, 2024-09-19:
Hi @henrykvdb ,

Even though it seems odd this is how the LRM specifies the declaration of parameters (A.2.1.1). This results in the behavior you get, i.e. all declaration nodes include the colon except for parameters and local parameters.

However it should not be an issue to obtain a correct syntax tree since the colon is included in higher levels of the tree. For example: 
```js
  class_item: $ => choice(
    seq(repeat($.attribute_instance), choice(
      $.class_property,
      $.class_method,
      $._class_constraint,
      $.class_declaration,
      $.interface_class_declaration,
      $.covergroup_declaration,
    )),
    seq($.any_parameter_declaration, ';'),
    ';',
    $._directives, // Out of LRM
  ),
```

gmlarumbe, 2024-09-19:
> My gut feel is telling me they omitted the ; so the definition can be reused between parameters passed to a module (comma separated) and parameters local to a module (followed by ;).

Yep, I agree that seems to be the reason.

On the other hand the snippet you posted should give an error:
```
(source_file [0, 0] - [4, 0]
  (module_declaration [0, 0] - [3, 9]
    (module_ansi_header [0, 0] - [0, 12]
      (module_keyword [0, 0] - [0, 6])
      name: (simple_identifier [0, 7] - [0, 11]))
    (local_parameter_declaration [1, 4] - [1, 28]
      (data_type_or_implicit [1, 15] - [1, 18]
        (data_type [1, 15] - [1, 18]
          (integer_atom_type [1, 15] - [1, 18])))
      (list_of_param_assignments [1, 19] - [1, 28]
        (param_assignment [1, 19] - [1, 28]
          (simple_identifier [1, 19] - [1, 24])
          (constant_param_expression [1, 27] - [1, 28]
            (constant_mintypmax_expression [1, 27] - [1, 28]
              (constant_expression [1, 27] - [1, 28]
                (constant_primary [1, 27] - [1, 28]
                  (primary_literal [1, 27] - [1, 28]
                    (integral_number [1, 27] - [1, 28]
                      (decimal_number [1, 27] - [1, 28]
                        (unsigned_number [1, 27] - [1, 28])))))))))))
    (comment [1, 29] - [1, 42])
    (ERROR [2, 4] - [2, 20]
      (simple_identifier [2, 4] - [2, 7])
      (simple_identifier [2, 8] - [2, 17])
      (decimal_digit [2, 19] - [2, 20]))))
```

> I still believe having every statement at least wrapped by some treesitter node would useful/cleaner, instead of leaving the floating ;'s that only belong to the parent and not the statement that they are terminating

I also do, but my feeling is that doing so would complicate things since it would differ from what the LRM states. And as you mentioned before, the LRM is taking into account that parameter/localparam declarations inside classes, packages, modules, etc... are always followed by a semicolon.

henrykvdb, 2024-09-19:
> I also do, but my feeling is that doing so would complicate things since it would differ from what the LRM states. And as you mentioned before, the LRM is taking into account that parameter/localparam declarations inside classes, packages, modules, etc... are always followed by a semicolon.

Does the LRM state this explicitly somewhere? I only saw this through the examples (all of them). In A.2.1.1 indeed there is no semicolon as you pointed out.

My use xase

henrykvdb, 2024-09-19:
While I see the value in sticking as closely as possible to the LRM, I still believe the primary goal of a tree-sitter should be to make a usable syntax tree that aids development.

Changing `local-param-declaration` to account for the necessary ; in modules is a bit dirty as then you diverge slightly from the LRM. Perhaps a cleaner solution would be to add a thin wrapper node around `local-param-declaration` that accounts for this ;``?

For me personally my use case for tree-sitter is highlighting and code navigation/modification based on the tree. For error checking an LSP seems like a more appropriate tool (although I guess having an ERROR in the tree is fair when the code is wrong)

gmlarumbe, 2024-09-19:
> Does the LRM state this explicitly somewhere? I only saw this through the examples (all of them). In A.2.1.1 indeed there is no semicolon as you pointed out.

You can check section 6.20.1 (Parameter declaration syntax). You will notice that only the `specparam_declaration` includes explicitly the semicolon. 

For an explicit non-usage of the semicolon you can check section A.1.3 (parameter_port_declaration). For explicit usage of the semicolon check sections A.1.5 (config_declaration), A.1.9 (class_item/interface_class_item), A.1.11 (package_or_generate_item_declaration) or A.2.8 (block_item_declaration).

> Changing local-param-declaration to account for the necessary ; in modules is a bit dirty as then you diverge slightly from the LRM. Perhaps a cleaner solution would be to add a thin wrapper node (e.g. local-param-declaration-in-module) around local-param-declaration that accounts for this ;?

Right now the grammar is written so that it sticks as close as possible to the LRM to make it simpler. The only deviations are to support some basic preprocessing capabilities or to constraint some specific cases that resulted in many conflicts while still providing a syntax tree whose node names correspond to LRM symbol names. 

Adding node `local_declaration_in_module` would probably simplify your tool workflow at the expense of making the grammar more complex (it would likely have an effect on how conflicts are handled). Therefore I believe this is something that should be taken into account by your specific tool rather than provided by the grammar itself.

> For me personally my use case for tree-sitter is highlighting and code navigation/modification based on the tree. For example I added a keybind to remove all children from a begin/end block, which can not work if the ; is not part of any child.

I am not aware of how to implement this with Nvim but this should not really be an issue as long as your tool shows anynomous nodes as part of the CST. In the case of parameters the semicolon in modules would simply be a sibling node of the parameter declaration node.

> For actual error checking an LSP seems like a more appropriate tool (although I guess having an ERROR in the tree is fair when the code is wrong)

LSP is probably more robust. However I have been using it for a while as a syntax checker and it works like a charm.
[image]

henrykvdb, 2024-09-19:
I don't really think it's a tool issue, for me the sensible situation would be that  for any parent node:
- the children are disjunct
- the union of the children equal the parent  (besides whitespaces and disregarding leaf nodes and whitespaces)

Now effectively the `;` is not accounted for by any of the children but only the parent. So walking through the code you will encounter the following

```
parent-A
child-of-parentA-pre-1
...
child-of-parentA-pre-N
parent-A
child-of-parentA-post-1
...
child-of-parentA-post-N
parent-A
```

Seems very strange to me?

Maybe another possible solution could be to add the `;` as a new node, as is done for unnecessary trailing `;`'s?

[image]

gmlarumbe, 2024-09-20:
Hi @henrykvdb ,

I am not sure I understand what your issue exactly is. I might be wrong but I think that you are not able to see the semicolon as an actual node in the syntax tree.

Parameters and localparams are defined in the grammar as a sequence of two nodes:
```js
    seq($.any_parameter_declaration, ';'),
```
That means that the semicolon is actually a node in the syntax tree (an anonymous node, see https://tree-sitter.github.io/tree-sitter/using-parsers#named-vs-anonymous-nodes). 

For previous case (e.g. a parameter that is a class item), the parent node is `class_item` and it has two child nodes: `parameter_declaration` and `;` (because `any_parameter_declaration` is inlined).

In the following example:
```verilog
localparam foo = 0;
```
My tool (Emacs) provides the following syntax tree (showing anonymous nodes):
```
(source_file
 (local_parameter_declaration localparam
  (list_of_param_assignments
   (param_assignment (simple_identifier) =
    (constant_param_expression
     (constant_mintypmax_expression
      (constant_expression
       (constant_primary
	(primary_literal
	 (integral_number
	  (decimal_number (unsigned_number)))))))))))
 ;)
```

You can see that it is correct syntax and that `local_parameter_declaration` node is a sibling of `;` (which is the expected behavior).

henrykvdb, 2024-09-20:
> I am not sure I understand what your issue exactly is. I might be wrong but I think that you are not able to see the semicolon as an actual node in the syntax tree.

Hi @gmlarumbe , I can't see the anonymous nodes indeed and it seemed to break my query for some reason. I made some changes to the query and found a workaround..

I still disagree and believe the primary goal of a parser should be to provide useful grammar to aid development, rather than having as primary goal to copy paste a spec (even where it does not make sense). In any case it seems like you made up your mind so I will close this issue.

closed 2024-09-20 by henrykvdb

gmlarumbe, 2024-09-21:
Hi @henrykvdb ,

> I still disagree and believe the primary goal of a parser should be to provide useful grammar to aid development, rather than having as primary goal to copy paste a spec (even where it does not make sense).

As I mentioned before I agree with you but there are some tradeoffs.

The reason to stick as close as possible to the LRM is to simplify the grammar and "copy/pasting the spec" results in conforming to the standard more easily. The grammar for SystemVerilog is already large and complex enough compared to the rest of tree-sitter grammars and it takes some time to debug it.

If you want to add new changes for the sake of your use case you need to take this into account and consider that adding a new 'thin wrapper node' is probably not as simple as you may think. 

- First thing would be considering sensible names. For example, if you call that new node `local-param-declaration-in-module` (as you suggested), what about the localparams in packages, configs, interfaces, classes, interface classes...? Would you create a different node for each of them? What names would you give them? Would you also rename `parameter_declaration` node in module parameter lists so that syntax tree would 'make more sense'? 
- Second thing, how would you handle new grammar conflicts? How would you rewrite the already existing ones?
- Finally, would this really solve an issue? How much time would be needed for such a change?

> In any case it seems like you made up your mind so I will close this issue.

I am more than open to collaborations, improvements and suggestions. 

If you are really interested in this enhancement you can take a look at how other tool parser generators (Verilator, Icarus, Verible...) handle this specific case and open a PR after solving conflicts and updating all the tests.