gmlarumbe/tree-sitter-systemverilog #7: node-field-name change AST

zoomer-k, 2024-11-21:
tree-sitter has field function
https://tree-sitter.github.io/tree-sitter/creating-parsers#using-fields

when generating AST only contains field name.

example

systemverilog
```
module dff ( input  logic clk); endmodule
```

use tree-sitter-systemverilog
```
source_file (0,0)-(0,41)
  module_declaration (0,0)-(0,41)
    module_ansi_header (0,0)-(0,31)
      module_keyword (0,0)-(0,6)
      name: simple_identifier (0,7)-(0,10)
      list_of_port_declarations (0,11)-(0,30)
        ansi_port_declaration (0,13)-(0,29)
          variable_port_header (0,13)-(0,25)
            port_direction (0,13)-(0,18)
            variable_port_type (0,20)-(0,25)
              data_type (0,20)-(0,25)
                integer_vector_type (0,20)-(0,25)
          port_name: simple_identifier (0,26)-(0,29)
```

use tree-sitter-verilog

```
source_file (0,0)-(0,41)
  module_declaration (0,0)-(0,41)
    module_header (0,0)-(0,10)
      module_keyword (0,0)-(0,6)
      simple_identifier (0,7)-(0,10)
    module_ansi_header (0,11)-(0,30)
      list_of_port_declarations (0,11)-(0,30)
        ansi_port_declaration (0,13)-(0,29)
          variable_port_header (0,13)-(0,25)
            port_direction (0,13)-(0,18)
            data_type (0,20)-(0,25)
              integer_vector_type (0,20)-(0,25)
          port_identifier (0,26)-(0,29)
            simple_identifier (0,26)-(0,29)
```

when using filed name , ommit port_identifier

I think port_identifier is need to close LRM

gmlarumbe, 2024-12-04:
Hi @zoomer-k ,

Thanks for opening an issue.

This is the expected AST since `tree-sitter-systemverilog` inlines all the identifiers to simplify conflict handling whereas `tree-sitter-verilog` inlines only a subset of these.

I agree that having `port_identifier` as a node could be useful in a query to get the identifier. That is the reason why the field `port_name` was implemented to retrieve the identifier of the port.

gmlarumbe, 2024-12-04:
Closing, feel free to reopen if needed.

closed 2024-12-04 by gmlarumbe