gmlarumbe/tree-sitter-systemverilog #10: ifndef preprocessing macro not detected

MinhTran0911, 2024-12-10:
While \`ifdef, \`elsif, \`endif works fine, the \`ifndef preprocessing macro is not parsed. 

Output of Inspect:
[image]

gmlarumbe, 2024-12-10:
Hi @MinhTran0911 ,

Thanks for opening an issue.

The following snippet works fine for me:
```verilog
`ifdef FOO_1
`elsif FOO_2
`endif
`ifndef FOO_1
`endif  
```

Parsed tree:
```
(source_file
 (conditional_compilation_directive directive_ifdef
  (ifdef_condition (simple_identifier)))
 (conditional_compilation_directive directive_elsif
  (ifdef_condition (simple_identifier)))
 (conditional_compilation_directive directive_endif)
 (conditional_compilation_directive directive_ifndef
  (ifdef_condition (simple_identifier)))
 (conditional_compilation_directive directive_endif))
```

Can you show the output of the command `$ tree-sitter-parse <filename>.sv` ?

MinhTran0911, 2024-12-11:
I tried with your snippet and get this parsed tree:
```
(source_file [0, 0] - [5, 0]
  (conditional_compilation_directive [0, 0] - [0, 12]
    (ifdef_condition [0, 7] - [0, 12]
      (simple_identifier [0, 7] - [0, 12])))
  (conditional_compilation_directive [1, 0] - [1, 12]
    (ifdef_condition [1, 7] - [1, 12]
      (simple_identifier [1, 7] - [1, 12])))
  (conditional_compilation_directive [2, 0] - [2, 6])
  (conditional_compilation_directive [3, 0] - [3, 13]
    (ifdef_condition [3, 8] - [3, 13]
      (simple_identifier [3, 8] - [3, 13])))
  (conditional_compilation_directive [4, 0] - [4, 6]))
```

gmlarumbe, 2024-12-11:
Hi @MinhTran0911 ,

I would say that the `ifndef` directive is being parsed correctly. What is the result that you would expect in the parsed tree?

MinhTran0911, 2024-12-11:
Hi @gmlarumbe,
I think there is no problem with the parsed tree. The reason I'm not seeing highlight for `ifndef` is because the query file of nvim-treesitter is missing `directive_ifndef`. Sorry for this.

And just out of curiosity, how do I generate the parsed tree in the format similar to your previous comment? I think query file uses those instead of the one I generated from `tree-sitter parse`, it would easier for me to edit the query file.

closed 2024-12-11 by MinhTran0911

gmlarumbe, 2024-12-11:
I generated that format with Emacs `treesit-explore-mode`. I am not familiar with nvim but there should probably be a similar tool to navigate the parsed syntax tree.

You can also try with `$ tree-sitter parse -h` but I think there is no such an option.

zhangwwpeng, 2024-12-12:
I am responsible for maintaining the verilog tree-sitter query in neovim. If there is anything missing in the query, please @ me and I will submit a PR to tree-sitter soon.

MinhTran0911, 2024-12-12:
Hi @zhangwwpeng,

Thank you for maintaining treesitter query for V/SV. I found that the current highlights.scm file is missing the `directive_ifndef`, that's why the `ifndef` macro is not highlighted properly.

[image]