gmlarumbe/tree-sitter-systemverilog #12: comment has white space in end of line

zoomer-k, 2024-12-13:
```
//test1;
//test2;
```

first line parse
`//test1; `
second line parse
`//test2;`

is this the expected behavior?

gmlarumbe, 2024-12-13:
Hi @zoomer-k ,

I am not sure what the issue is here. 

The output of `$ tree-sitter parse <file.sv>` gives the following parsed tree, which seems correct to me:
```
(source_file [0, 0] - [2, 0]
  (comment [0, 0] - [0, 8])
  (comment [1, 0] - [1, 8]))
```

Please first make sure it is not a tool related issue instead of a parser error running the `tree-sitter parse` command on the snippet.

zoomer-k, 2024-12-16:
Hi @gmlarumbe 
thank you for researching
i try it in windows10

```
(source_file [0, 0] - [1, 8]
  (comment [0, 0] - [0, 9])
  (comment [1, 0] - [1, 8])) 
```

gmlarumbe, 2024-12-16:
Hi @zoomer-k ,

It seems it might be because of CRLF Windows format for newlines.

Can you try updating tree-sitter to the latest release and check the result of the parsed tree?

Thanks!

zoomer-k, 2024-12-17:
Hi @gmlarumbe 
I'm trying but recently tree-sitter add tree-sitter.json file
https://github.com/tree-sitter/tree-sitter/issues/3637

could you update repository for latest tree-sitter?

gmlarumbe, 2024-12-21:
Hi @zoomer-k ,

I updated generated files with tree-sitter 0.24 on 4344f7e, including the new `tree-sitter.json` file. 

Hope it helps!

closed 2025-01-08 by gmlarumbe