gmlarumbe/tree-sitter-systemverilog #13: In SystemVerilog, macro definition parsing error

donniek, 2024-12-20:
[image]

It seems there has been a misunderstanding in the translation request. The original text you provided is already in English. However, I believe you are asking for assistance with a SystemVerilog parsing issue where a conditional compilation directive like `ifdef is causing an error.
Thanks.

donniek, 2024-12-21:
--- a/tree-sitter-verilog/grammar.js
+++ b/tree-sitter-verilog/grammar.js
@@ -489,7 +489,8 @@ const rules = {
     '(',
     commaSep(seq(
       repeat($.attribute_instance),
-      $.ansi_port_declaration)
+      seq(optional($._directives), $.ansi_port_declaration)
+      ),
     ),
     ')'
   ),
It appears that this was a small problem, and I have already addressed the processing of macro definitions by making the aforementioned modifications.

gmlarumbe, 2024-12-21:
Hi @donniek ,

Thanks for reporting an issue. To me this looks a duplicate of #11. 

Thanks a lot for taking the time to find a solution too. It is similar to the one I proposed at https://github.com/gmlarumbe/tree-sitter-systemverilog/issues/11#issuecomment-2538978371. Could you check if it works for you and open an PR with a new testcase?

To create a new testcase, create a file under `test/files/core/directives` (e.g. module_port_list.sv) and run:
`$ ./update_tests.sh -t -p core/directives/module_port_list`

This will create a new reference test file (module_port_list.txt) that you can add to the PR along with the module_port_list.sv file.

donniek, 2024-12-22:
I'm going to give it a try following the process you outlined. 
I'm truly grateful for your swift response. I must say, I'm thoroughly impressed by your work; the tree-sitter-systemverilog project you've developed is incredibly thorough and high-quality. It certainly warrants more recognition. It's great to see that it's now integrated into Emacs verilog-ts-mode and nvim systemverilog, which should certainly help it earn more stars. 
Keep up the outstanding work!💪

> Hi @donniek ,
> 
> Thanks for reporting an issue. To me this looks a duplicate of #11.
> 
> Thanks a lot for taking the time to find a solution too. It is similar to the one I proposed at [#11 (comment)](https://github.com/gmlarumbe/tree-sitter-systemverilog/issues/11#issuecomment-2538978371). Could you check if it works for you and open an PR with a new testcase?
> 
> To create a new testcase, create a file under `test/files/core/directives` (e.g. module_port_list.sv) and run: `$ ./update_tests.sh -t -p core/directives/module_port_list`
> 
> This will create a new reference test file (module_port_list.txt) that you can add to the PR along with the module_port_list.sv file.

gmlarumbe, 2024-12-24:
Fixed in #14

closed 2024-12-24 by gmlarumbe