gmlarumbe/tree-sitter-systemverilog #17: Fails to parse size-casting

numero-744, 2025-03-10:
Hello,

I was pointed to this repository from https://github.com/tree-sitter/tree-sitter-verilog/issues/81
Thanks for the your great work!

I would like to add SystemVerilog support to [`mergiraf`](https://codeberg.org/mergiraf/mergiraf/releases/tag/v0.6.0), which uses tree-sitter.
I had tried `tree-sitter-verilog`, which has many issues.
Now I am trying this repository, which is better, but I still have issues with code from CVA6 repository.
Well, CVA6 looks like an interesting playground for integration testing 😉

The first error appears in https://github.com/openhwgroup/cva6/blob/master/core/scoreboard.sv
The `scoreboard.sv` file is correctly parsed by verilator, questa and vcs, but not by this repository 🤔

Here is the first error:

```
% tree-sitter parse --scope source.verilog core/scoreboard.sv
...
core/scoreboard.sv	Parse:    5.13 ms	 2989 bytes/ms	(ERROR [201, 18] - [201, 20])
% tree-sitter langs
scope: source.verilog
parser: "/home/come/Development/tree-sitter/tree-sitter-systemverilog/."
highlights: None
file_types: []
content_regex: None
injection_regex: Some(Regex("^verilog$"))
```

Here is line 202, column 19:

```
            mem_n[8'(trans_id_i[i])-1].sbe.valid = 1'b1;
                   ^ this type casting operation seems to trigger a parse error
```

gmlarumbe, 2025-03-10:
Hi @numero-744 ,

Thanks for reporting the issue. 

The `$.integral_number` choice in `$.casting_type` was missing in the grammar by mistake. It has been fixed in 23df90f and d589921. If you push the latest commit it should work.

Let me know if it does not and feel free to reopen the issue in that case.

Thanks!

closed 2025-03-10 by gmlarumbe

numero-744, 2025-03-10:
Thank you very much for the quick fix! 🙏
You didn't lie about maintenance 😃 

I have other problems in other files, I'll try to isolate them before opening issues (probably before the end of the week).

See you later 👋

drom, 2025-07-10:
@gmlarumbe I would appreciate PR to this repo instead of redirecting to unrelated project.