gmlarumbe/tree-sitter-systemverilog #19: Question: be less restrictive with type casting?

numero-744, 2025-03-12:
I have found this: https://github.com/openhwgroup/cva6/issues/2824

To put it short, the standard seems to prevent things such as `Cfg.Len'(value)` in the syntax.
(At least I have not found where it is allowed.)
The type can be `42`, `unsigned`, `logic`, `my_t`, `my_pkg::my_t` but not `Cfg.Len`.
(This is only a syntax issue, `Cfg` is a parameter so semantically, it is equivalent to `typedef logic [Cfg.Len-1:0] my_t;` then `my_t'(value)`).

However, Verilator, Verible, Questa, Vcs, Vivado (and probably others) allow `Cfg.Len'(value)`.

Should we support this syntax as well?

gmlarumbe, 2025-03-14:
Should be supported after [44f8228](https://github.com/gmlarumbe/tree-sitter-systemverilog/commit/44f8228dea44ca13333929a28c5a5bdcb22eb10f).

Feel free to reopen if I missed something.

closed 2025-03-14 by gmlarumbe

numero-744, 2025-03-15:
Thanks again! This change significantly reduced the number of errors while parsing the whole CVA6 `core/` directory 😃