gmlarumbe/tree-sitter-systemverilog #21: add package-lock.json

zoomer-k, 2025-03-24:
Hi

thank you for your great repository

it is recommended keep lock file in repository

https://github.com/actions/setup-node/blob/main/docs/advanced-usage.md#working-with-lockfiles

could you add it?

and, it is need for checking test binding in node

https://github.com/tree-sitter/tree-sitter/issues/4234

node-tree-sitter is not yet update to v0.25.3 

https://github.com/tree-sitter/node-tree-sitter/pull/241

gmlarumbe, 2025-03-24:
Hi @zoomer-k ,

The grammar has not been published yet to crates, npm or PyPi. There are some guidelines about how to proceed in the tree-sitter doc: https://tree-sitter.github.io/tree-sitter/creating-parsers/6-publishing.html

I plan to spend some time on this task once I have fixed the issue with the ABI version in the sv-tests repo.

Thanks for your feedback and all your help!

gmlarumbe, 2025-03-25:
I tried using the `publish.yml` GitHub workflow for npm similar to that one of `tree-sitter-python`. However the npm build for Windows gave the following error:
```
binding.cc
  parser.c
  win_delay_load_hook.cc
     Creating library D:\a\tree-sitter-systemverilog\tree-sitter-systemverilog\build\Release\tree_sitter_systemverilog_binding.lib and object D:\a\tree-sitter-systemverilog\tree-sitter-systemverilog\build\Release\tree_sitter_systemverilog_binding.exp
binding.obj : error LNK2001: unresolved external symbol tree_sitter_systemverilog [D:\a\tree-sitter-systemverilog\tree-sitter-systemverilog\build\tree_sitter_systemverilog_binding.vcxproj]
D:\a\tree-sitter-systemverilog\tree-sitter-systemverilog\build\Release\tree_sitter_systemverilog_binding.node : fatal error LNK1120: 1 unresolved externals [D:\a\tree-sitter-systemverilog\tree-sitter-systemverilog\build\tree_sitter_systemverilog_binding.vcxproj]
gyp ERR! build error 
gyp ERR! stack Error: `C:\Program Files\Microsoft Visual Studio\2022\Enterprise\MSBuild\Current\Bin\MSBuild.exe` failed with exit code: 1
gyp ERR! stack at ChildProcess.<anonymous> (C:\hostedtoolcache\windows\node\20.18.3\x64\node_modules\npm\node_modules\node-gyp\lib\build.js:209:23)
gyp ERR! stack at ChildProcess.emit (node:events:518:28)
gyp ERR! stack at ChildProcess._handle.onexit (node:internal/child_process:293:12)
gyp ERR! System Windows_NT 10.0.20348
gyp ERR! command "C:\\hostedtoolcache\\windows\\node\\20.18.3\\x64\\node.exe" "C:\\hostedtoolcache\\windows\\node\\20.18.3\\x64\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd D:\a\tree-sitter-systemverilog\tree-sitter-systemverilog
gyp ERR! node -v v20.18.3
gyp ERR! node-gyp -v v10.1.0
gyp ERR! not ok 
npm error code 1
npm error path D:\a\tree-sitter-systemverilog\tree-sitter-systemverilog
npm error command failed
npm error command C:\Windows\system32\cmd.exe /d /s /c node-gyp-build
npm error A complete log of this run can be found in: C:\npm\cache\_logs\2025-03-25T17_35_27_347Z-debug-0.log
```

Any idea of why could this be happening?

zoomer-k, 2025-03-27:
@gmlarumbe 

thankyou for reply , ill check it

i tried it to use [tree-sitter action ](https://github.com/tree-sitter/parser-test-action) 

cant pass node bind test in [my repository](https://github.com/zoomer-k/tree-sitter-systemverilog)

tree-sitter-verilog can pass test in [my repository](https://github.com/zoomer-k/tree-sitter-verilog/actions/runs/14029060356)
and other env (pip go and so on)

zoomer-k, 2025-03-27:
@gmlarumbe 
I try it in tree-sitter-verilog

https://github.com/zoomer-k/tree-sitter-verilog/actions/runs/14100529012

could pass

gmlarumbe, 2025-04-01:
Hi @zoomer-k , 

Thanks for your help tracing back the issue. It seems it has to do with the different names of `verilog` in the grammar module.exports and `systemverilog` in the repo.

```js
module.exports = grammar({
  name: 'verilog',
  word: $ => $.simple_identifier,
  rules: rules,
  extras: $ => [/\s/, $.comment],
```

However the name `verilog` is assumed by tools like Emacs so it cannot be changed. On the other hand it only happens in Windows builds. 

Seems this could be an issue of `tree-sitter` and we could open a ticket in that repo.

zoomer-k, 2025-06-27:
thank you for your cooperation
tree-sitter-aciton may have another problem.... 

I'll check it

zoomer-k, 2025-07-04:
Hi @gmlarumbe 
I checked official bind CI/CD .
but couldnt finish 
https://github.com/zoomer-k/tree-sitter-systemverilog/actions/runs/15891748143/job/44815436211

go , swift , node have some problem

nodes error code

```
found 0 vulnerabilities
Run node --test bindings/node/*_test.js
  node --test bindings/node/*_test.js
  shell: /usr/bin/sh -e {0}
  env:
    TREE_SITTER_ABI_VERSION: 14
    TREE_SITTER_REF: v0.25.6
    TREE_SITTER_REPO: https://github.com/tree-sitter/tree-sitter
    pythonLocation: /opt/hostedtoolcache/Python/3.12.10/x64
    PKG_CONFIG_PATH: /opt/hostedtoolcache/Python/3.12.10/x64/lib/pkgconfig
    Python_ROOT_DIR: /opt/hostedtoolcache/Python/3.12.10/x64
    Python2_ROOT_DIR: /opt/hostedtoolcache/Python/3.12.10/x64
    Python3_ROOT_DIR: /opt/hostedtoolcache/Python/3.12.10/x64
    LD_LIBRARY_PATH: /opt/hostedtoolcache/Python/3.12.10/x64/lib
    CARGO_INCREMENTAL: 0
    CARGO_PROFILE_DEV_DEBUG: 0
    CARGO_TERM_COLOR: always
    RUST_BACKTRACE: short
    RUSTFLAGS: -D warnings
    CARGO_UNSTABLE_SPARSE_REGISTRY: true
    CARGO_REGISTRIES_CRATES_IO_PROTOCOL: sparse
    CACHE_ON_FAILURE: false
TAP version 13
# node:internal/modules/cjs/loader:1404
#   throw err;
#   ^
# Error: Cannot find module 'tree-sitter'
# Require stack:
# - /home/runner/work/tree-sitter-systemverilog/tree-sitter-systemverilog/bindings/node/binding_test.js
#     at Function._resolveFilename (node:internal/modules/cjs/loader:1401:15)
#     at defaultResolveImpl (node:internal/modules/cjs/loader:1057:19)
#     at resolveForCJSWithHooks (node:internal/modules/cjs/loader:1062:22)
#     at Function._load (node:internal/modules/cjs/loader:1211:37)
#     at TracingChannel.traceSync (node:diagnostics_channel:322:14)
#     at wrapModuleLoad (node:internal/modules/cjs/loader:235:24)
#     at Module.require (node:internal/modules/cjs/loader:1487:12)
#     at require (node:internal/modules/helpers:135:16)
#     at Object.<anonymous> (/home/runner/work/tree-sitter-systemverilog/tree-sitter-systemverilog/bindings/node/binding_test.js:4:16)
#     at Module._compile (node:internal/modules/cjs/loader:1730:14) {
#   code: 'MODULE_NOT_FOUND',
#   requireStack: [
#     '/home/runner/work/tree-sitter-systemverilog/tree-sitter-systemverilog/bindings/node/binding_test.js'
#   ]
# }
# Node.js v22.16.0
# Subtest: bindings/node/binding_test.js
not ok 1 - bindings/node/binding_test.js
  ---
  duration_ms: 33.408081
  type: 'test'
  location: '/home/runner/work/tree-sitter-systemverilog/tree-sitter-systemverilog/bindings/node/binding_test.js:1:1'
  failureType: 'testCodeFailure'
  exitCode: 1
  signal: ~
  error: 'test failed'
  code: 'ERR_TEST_FAILURE'
  ...
```

sorry I dont have time to search it

gmlarumbe, 2025-07-08:
Moving to a new issue.

closed 2025-07-08 by gmlarumbe