gmlarumbe/tree-sitter-systemverilog #26: Compatibility with py-tree-sitter

tafk7, 2025-05-15:
I'm attempting to leverage this grammar with via the `py-tree-sitter` python package, but am having multiple issues:
1. Error about the grammar name in `parser.c`. Since py-tree-sitter removed directly loading grammars from `.so` files in v23, I need to load it like [this](https://github.com/tree-sitter/py-tree-sitter/discussions/251) which errors if the grammar name isn't consistently systemverilog. I understand the emacs use case [expects "verilog" ](https://github.com/gmlarumbe/tree-sitter-systemverilog/issues/21#issuecomment-2769498141) but it seems like a bad idea to try and maintain the wrong name since it's causing issues (and not just from windows systems, this error with py-tree-sitter happens on Linux): 
- #21 
- #22
- #25 
2. Error about the language version. The most recent py-tree-sitter release uses tree-sitter version 24 and thus expects v14 compilation from the CLI. However, even when compiled targeting v14 from the proper versioned tools, the grammar is marked as v15 and fails to be loaded. I was able to fix the problem by rolling back to [Commit 6da2084](https://github.com/gmlarumbe/tree-sitter-systemverilog/commit/6da2084737712dea955b9ca1bf5261b5621dad18), but would like this to be fixed on main. 
- Similar issue: #23 

Fixes to both these issues and/or more support for use with `py-tree-sitter` would be extremely appreciated! This grammar would be very valuable for my project.

gmlarumbe, 2025-06-23:
Hi @tafk7 ,

Sorry for the late response and thanks for taking your time to debug these issues:

>Error about the grammar name in parser.c. Since py-tree-sitter removed directly loading grammars from .so files in v23, I need to load it like https://github.com/tree-sitter/py-tree-sitter/discussions/251 which errors if the grammar name isn't consistently systemverilog. I understand the emacs use case https://github.com/gmlarumbe/tree-sitter-systemverilog/issues/21#issuecomment-2769498141but it seems like a bad idea to try and maintain the wrong name since it's causing issues (and not just from windows systems, this error with py-tree-sitter happens on Linux):

This has been recently fixed in #29. Let me know if everything works well for your application.

> Error about the language version. The most recent py-tree-sitter release uses tree-sitter version 24 and thus expects v14 compilation from the CLI. However, even when compiled targeting v14 from the [...]

Support for ABI 15 was recently merged to master in `py-tree-sitter`: https://github.com/tree-sitter/py-tree-sitter/pull/333

There is still no release that includes these changes. However I hope that you can fetch the code from master.

Thanks a lot!

gmlarumbe, 2025-07-14:
Maybe you can give a try to the link below until original repo provides a release: 
- https://pypi.org/project/tree-sitter-language-version-15/

gmlarumbe, 2025-07-29:
Tested locally, seems to work with v0.25.0 of `py-tree-sitter`.

```python
#!/usr/bin/env python3

import tree_sitter_systemverilog as systemverilog
from tree_sitter import Language, Parser

SV_LANGUAGE = Language(systemverilog.language())

parser = Parser(SV_LANGUAGE)

src = bytes(
"""
module foo ();
endmodule
""",
        "utf8"
    )

def read_callable_byte_offset(byte_offset, point):
    return src[byte_offset : byte_offset + 1]

tree = parser.parse(read_callable_byte_offset, encoding="utf8")

print(str(tree.root_node))
```

Result:
```
(source_file (module_declaration (module_nonansi_header (module_keyword) name: (simple_identifier) (list_of_ports))))
```

closed 2025-07-29 by gmlarumbe