gmlarumbe/tree-sitter-systemverilog #35: Parsing error in expression with method calls within part select

gmlarumbe, 2025-07-11:
```verilog
a.a = b[i][A.a()+B.b()-1:0]; // ERROR

// The ones below are OK
a.a = b[i][A()+B()-1:0];
a.a = b[i][A.a+B.b-1:0];
a.a = b[i][A+B-1:0];
```

lwflwf1, 2026-09-28:
The refined approach in #68 also fixes this case. With `method_call` split so that `::` requires parentheses, plus `bit_select` accepting `[constant_range]`, this parses without `ERROR`:

```verilog
a.a = b[i][A.a()+B.b()-1:0];
```

(it is an `ERROR` on master/v0.4.1). Neither change alone is enough: removing the `bit_select` part brings the parse error back.

Regression: `./update_tests.sh -t` in my fork passes with **0 files gaining `ERROR`/`MISSING`** (both standalone and merged on top of master). See #68 for the full diff.

gmlarumbe, 2026-10-21:
Fixed with #86

closed 2026-10-21 by gmlarumbe