gmlarumbe/tree-sitter-systemverilog #36: vim-matchup(https://github.com/andymass/vim-matchup) and tree-sitter-systemverilog are incompatible in neovim

Aprilskysky, 2025-07-20:
when enable verilog/systemverilog treessiter parser, vim-matchup will not work.

gmlarumbe, 2025-07-20:
Hi @Aprilskysky ,

For tool-specific related issues you should contact the tool developers/maintainers.

However feel free to reopen if after some debugging you find out it has to do with the parser.

closed 2025-07-20 by gmlarumbe

gmlarumbe, 2025-07-20:
I am afraid I cannot help you since I am not a Vim/Neovim user. Besides the issue you describe seems an integration error of the grammar in the tool, not an issue on the grammar itself.

AFAIK Neovim tree-sitter makes use of this parser and not of `tree-sitter-verilog`. Maybe I would ask the developers/maintainers of `vim-matchup` to double check that is also the case for their plugin.

Aprilskysky, 2025-07-21:
Thank you very much for your efforts to help.